patjak/mba6x_bl #24: Starting Mac in Ubuntu after shut down from Ubuntu=No problem. Starting Mac in Ubuntu after shutdown from OSX=Dark Screen

hpe009, 2015-04-01:
I am using refind boot-menu, when I start up the computer after shutting down from ubuntu there is no problem after starting up in Ubuntu. But when i start up the computer after I have been in OS X, I get a black screen when the Mac is started up in Ubuntu. I have the mba6x_bl kernel module installed. this is what happen with me.

patjak, 2015-04-01:
I have the same problem but instead of a black screen I get the i915 default backlight functionality. I'm currently relocating and starting a new job but as soon as things calm down I'll look into it.

hpe009, 2015-04-03:
@patjak I know that this is of topic, but do you know how to make refind not show vmlinuz-kernelversion. And only show the grub bootloader as an option?

patjak, 2015-04-03:
After a firmware upgrade I couldn't run refind anymore so I put grub directly on the EFI partition and blessed it. Now I boot straight into Linux and if I need OSX I just hold down "option" which gives me the built-in boot menu. Hope that helps

hpe009, 2015-04-04:
I am a noob with that sort of thing, so you could not help me do that? @patjak

eugenecormier, 2015-04-04:
@hpe009 that is way off topic.... uninstall refind and using a ubuntu livecd (with efi) reinstall the bootloader... google is your friend

patjak, 2015-04-04:
Yes it's a bit off topic and to be honest I don't really remember what I did. I found a "note to self" post on G+ where I explained how to get Refind working after the SMC upgrade. I guess you can do the same but replace Refind with Grub and things should work. Here's what I did: https://plus.google.com/+PatrikJakobsson/posts/GvyJz21XsCM

hpe009, 2016-01-20:
Closing as I can not reproduce the problem anymore, and it may had something to do with Linux/Ubuntu

closed 2016-01-20 by hpe009