patjak/mba6x_bl #28: Issue when connected to power

petey33, 2015-06-08:
Hi,

How do I uninstall the mba6x_bl?

I have an issue when my mba is connected to the power cable. Then the machine will not suspend.

I thought maybe it had something to do with mba6x_bl? Just a wild guess.

Regards
Petey

rembo10, 2015-06-30:
I'm not sure if it's a mba6x_bl issue. Does it resume instantly after suspending? I was having the same issue even without mba6x_bl installed

pskoncej, 2015-08-07:
Hi,

Nevertheless, do you know how to uninstall mba6x_bl?

patjak, 2015-08-07:
@pskoncej if you compiled and installed from source you can just remove the kernel module. Think it is in /lib/modules/your-kernel-version/kernel/extra/mba6x_bl.ko
If you installed it with some package, just uninstall that package.

eugenecormier, 2015-09-23:
if the laptop won;t suspend, check:
cat /proc/acpi/wakeup
I got mine to suspend properly by disabling XHC1:
echo XHC1 > /proc/acpi/wakeup

petey33, 2015-09-28:
Thanks a lot eugenecormier.

Now the mba suspends almost every time. Sometimes, though, it freezes or won't turn off the screen when I close the lid. Do you have the same problems?

closed 2016-04-09 by patjak