patjak/mba6x_bl #32: Make error on custom kernel 4.2.0-3 (Arch)

frank604, 2015-09-09:
Commenting out the line resolves the make error and mba6x_bl still works after suspend/resume.  Is this due to my kernel config? 

I use the pkgbuild from testing repo in ABS and modify only kernel name and using modprobed-db.  Not sure if this info is needed but thought I'd give a heads up on this error just in case.  Love the work you do!

cat /var/lib/dkms/mba6x_bl/39.fc64de1/build/make.log                                                                                                                                                   

DKMS make.log for mba6x_bl-39.fc64de1 for kernel 4.2.0-3-mba (x86_64)
Tue Sep  8 11:52:27 PDT 2015
make -C /lib/modules/4.2.0-3-mba/build M=/var/lib/dkms/mba6x_bl/39.fc64de1/build modules
make[1]: Entering directory '/usr/lib/modules/4.2.0-3-mba/build'
  CC [M]  /var/lib/dkms/mba6x_bl/39.fc64de1/build/mba6x_bl.o
/var/lib/dkms/mba6x_bl/39.fc64de1/build/mba6x_bl.c: In function ‘platform_probe’:
/var/lib/dkms/mba6x_bl/39.fc64de1/build/mba6x_bl.c:336:2: error: implicit declaration of function ‘acpi_video_dmi_promote_vendor’ [-Werror=implicit-function-declaration]
  acpi_video_dmi_promote_vendor();
  ^
cc1: some warnings being treated as errors
scripts/Makefile.build:264: recipe for target '/var/lib/dkms/mba6x_bl/39.fc64de1/build/mba6x_bl.o' failed
make[2]: *** [/var/lib/dkms/mba6x_bl/39.fc64de1/build/mba6x_bl.o] Error 1
Makefile:1386: recipe for target '_module_/var/lib/dkms/mba6x_bl/39.fc64de1/build' failed
make[1]: *** [_module_/var/lib/dkms/mba6x_bl/39.fc64de1/build] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.2.0-3-mba/build'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2

patjak, 2015-09-10:
Thanks for letting me know. Seems they have reworked the backlight interface selection API. I will need to start using that. Will probably have to make a new branch for 4.2+ kernels.

eugenecormier, 2015-09-23:
yep, I hit this problem as well

PR #34 (2015-09-26): Support kernel >= 4.2
This should fix #32
closed 2015-09-26 by patjak