patjak/mba6x_bl #33: Doesn't work with Linux Mint running on Kernel 3.16 (Macbook Air mid-2014)

napping, 2015-09-16:
Hi,

I followed the instructions on my Macbook Air mid-2014, and when I turn on the mod, the screen just changes to a 30ish percent and stays there.  I try to change the brightness with the brightness keys, but it just remains this 30% brightness.  When I run rmmod, it goes back to the original issue, where it is at max brightness when I toggle it max brightness, and then goes black when I decreased the brightness to past 80%.  I've played around with some stuff like adding the xorg.conf, following the Ubuntuforums instructions, running make clean and make install again, etc., but nothing seems to be working.

What would you suggest I do?

sbng, 2015-09-29:
you can adjust the brightness via "echo X >  > /sys/class/backlight/mba6x_backlight/brightness" where X is the desired level of brightness.

napping, 2015-10-03:
Thanks, this does work for me.  However, is there a way for me to get the full brightness toggle functionality working using this driver?

sbng, 2015-10-04:
I am on 3.19.30 with i915 bpo. Macbook 7,2, works flawlessly. Maybe you like to give 3.19 a try. However, I did need to *get rid* of acpi_backlight=vendor from kernel boot option line to get this working.