patjak/mba6x_bl #35: Current deb package not working.

easdbe, 2015-09-28:
The prebuilt deb package (mba6xbl-dkms_1.0.0_all) doesn't install on 4.2.0-1-amd64.

DKMS make.log for mba6xbl-1.0.0 for kernel 4.2.0-1-amd64 (amd64)
Mon 28 Sep 06:01:01 BST 2015
make -C /lib/modules/4.2.0-1-amd64/build M=/var/lib/dkms/mba6xbl/1.0.0/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.2.0-1-amd64'
Makefile:10: *** mixed implicit and normal rules: deprecated syntax
  CC [M]  /var/lib/dkms/mba6xbl/1.0.0/build/mba6x_bl.o
/var/lib/dkms/mba6xbl/1.0.0/build/mba6x_bl.c: In function ‘platform_probe’:
/var/lib/dkms/mba6xbl/1.0.0/build/mba6x_bl.c:336:2: error: implicit declaration of function ‘acpi_video_dmi_promote_vendor’ [-Werror=implicit-function-declaration]
  acpi_video_dmi_promote_vendor();
  ^
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.2.0-1-common/scripts/Makefile.build:269: recipe for target '/var/lib/dkms/mba6xbl/1.0.0/build/mba6x_bl.o' failed
make[4]: *** [/var/lib/dkms/mba6xbl/1.0.0/build/mba6x_bl.o] Error 1
/usr/src/linux-headers-4.2.0-1-common/Makefile:1403: recipe for target '_module_/var/lib/dkms/mba6xbl/1.0.0/build' failed
make[3]: *** [_module_/var/lib/dkms/mba6xbl/1.0.0/build] Error 2
Makefile:146: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.2.0-1-amd64'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2
(END)

easdbe, 2015-09-28:
Never mind, the backlight appears to working just fine on a MacbookAir7,1 without mba6x_bl.

closed 2015-09-28 by easdbe

miekg, 2015-09-28:
[ Quoting <notifications@github.com> in "[mba6x_bl] Current deb package not ..." ]
>The prebuilt deb package (mba6xbl-dkms_1.0.0_all) doesn't install on 4.2.0-1-amd64.

Still should probably be updated for the latest kernels. Alas I'm only running
ubuntu 14.04, so I am not sure if I will be able.

/Miek

--
Miek Gieben

patjak, 2015-09-28:
For reference, I just merged a patch for >= 4.2 support though it's a compile time define so one .deb cannot support both pre 4.2 and newer.