patjak/mba6x_bl #36: mba6x_bl power consumption issue

sbng, 2015-09-29:
It appears that after loading the mba6x_bl module, powertop is seeing double the backlight power consumption. In my case, this reduces the battery power by about 20%. Is there any plans to optimize the mba6x_bl module to consume less power?

Power est.    Usage     Device name
  4.02 W     39.2%        Display backlight
  4.02 W     25.2%        Display backlight
  2.26 W    100.0%        USB device: USB Receiver (Logitech)
  1.50 W    100.0%        USB device: Apple Internal Keyboard / Trackpad (Apple
  652 mW      0.0 pkts/s  nic:virbr0

patjak, 2015-10-01:
I think powertop is just reporting it wrong because we have two backlight devices claiming to be active (i915 still thinks it's being used). If you would like to be 100% sure you can ofc measure how long your battery life lasts with and without mba6x_bl.

sbng, 2015-10-02:
Yes, I think you are correct. I don't see significant reduction in my power consumption. Thanks for the clarification.

closed 2015-10-15 by patjak