patjak/mba6x_bl #37: brightness alternating after install

winst0nsmith, 2015-10-14:
first: great work on this module, i've been using it for months no problems. thanks for the work!

i recently upgraded to debian sid, dkms rebuild failed. no sweat, uninstalled package, built from source after git clone. brightness issues after sleep are fixed as per usual.

now my brightness dims and brightens slightly of it's own accord. i don't see any adaptive brightness controls in gnome, so i can't figure out what is up. any hints?

also, i gather from other threads that powertop's reporting of the intel brightness module creating high cpu usage is a false report due to the way your module sets up. is that right? my brightness 'slow flicker' is not caused by these two module competing is it?

tia

patjak, 2015-10-14:
Not sure what's going on here but sounds like you're on to something. Either adaptive brightness is kicking in or the two drivers are racing. Though the i915 backlight should be disabled by mba6x_bl through a hardware switch but since the SMC keeps changing it's behaviour it's hard to say. You can test if i915 is getting in the way by testing if you manually can adjusting it's brightness in /sys/class/backlight/intel_backlight/brightness
If you can, something is wrong.

winst0nsmith, 2015-10-15:
thanks for the quick response.

changing the value in the intel file has no effect, while the number in the mba6x brightness file is constantly changing. looks like it is the mba6x module that is being adjusted by some mechanism.

running debian sid with kernel 4.2.0. any other info useful for troubleshooting?

winst0nsmith, 2015-10-15:
how to uninstall the module? modprobe -r mba6x_bl gives me a kernel panic, and of course reloads the module on the next boot. can i simply blacklist it in /etc/modules-load.d/ ?

i would like to uninstall completely to troubleshoot further.

patjak, 2015-10-15:
Hmm it shouldn't segfault. I think I need to run some tests on newer kernels...
You can just remove the file: /lib/modules/<kernel-version>/extra/mba6x_bl.ko
or you can blacklist it with kernel parameter: modprobe.blacklist=mba6x_bl
There are distro specific ways as well, use google for that.

agimenez, 2015-10-22:
Maybe the light sensor and some desktop/scripts trying to adapt your screen's brightness to the ambient light?

winst0nsmith, 2015-10-27:
yeah this is definitely as simple as the light sensor and something using it to set brightness. i wish i could figure out how to disable this behavior, i don't use it on any platform or os, and i sure don't want it if it's this flaky. google has been no friend to me on this one, looks like it'll have to be the decidedly low tech tactic: i'll stick some tape on the sensor...

agimenez, 2015-10-28:
I'm pretty sure you can disable it by using some echo 0 > /sys/some/path. What I can't check right now is what file is the one.

Using a quick search, I found this in the always fantastic ArchLinux Wiki entry:
https://wiki.archlinux.org/index.php/MacBook#Light_sensor

One of the links point to https://github.com/Janhouse/lighter/blob/master/lighter.pl, a simple script to handle brightness and ambient light, and I think that the file you want is:

```perl
# Path for light sensor.
my $sensor="/sys/devices/platform/applesmc.768/light";
```

Hope that helps!

patjak, 2016-04-09:
Closing as this is not mba6x_bl related.

closed 2016-04-09 by patjak