patjak/mba6x_bl #40: module no longer works with 4.2.0-18-generic #22-Ubuntu SMP

thekswenson, 2015-11-12:
Thanks for the module...
  I've been using it for a year a so.  It's integral to the enjoyment of my laptop.

Recently I've updated the kernel a few times and just noticed that the module doesn't seem to be working anymore.
After doing "sudo make install" and "sudo depmod -a" and "sudo modprobe mba6x_bl" and rebooting I see that the module is loaded:
"""
=> lsmod | grep mba
mba6x_bl               16384  0
video                  36864  2 i915,mba6x_bl
"""

Is it possible I'm doing something wrong?

patjak, 2015-11-15:
Hi, can you describe the problem? Is it still possible to change the backlight or is it just stuck at some level?

thekswenson, 2015-11-15:
Loading the module doesn't change anything.  After suspend, the brightness
can be all or nothing only, as if I hadn't loaded the module.

patjak, 2015-11-16:
Do you have a /sys/class/backlight/mba6x_bl directory when the module is loaded?

patjak, 2015-11-16:
Oh, and could you attach a full dmesg when mba6x_bl is being loaded?

thekswenson, 2015-11-16:
There is no /sys/class/backlight/mba6x_bl.
=> ls /sys/class/backlight/
acpi_video0@  intel_backlight@

thekswenson, 2015-11-16:
[  200.936799] mba6x_bl: Found LP8550 backlight driver

thekswenson, 2015-11-16:
I don't understand what is going on now but the process of loading and unloading (with modprobe, nothing else) the module has changed everything...
    things seem to be working normally for now!?

Also the contents of /sys/class/backlight/ have changed:
=> ls /sys/class/backlight/
intel_backlight@  mba6x_backlight@

patjak, 2015-11-16:
That's odd. Lets just pretend we fixed it and that nobody else runs into this :)

Reopen if the problem comes back.

closed 2015-11-16 by patjak

proninyaroslav, 2015-12-08:
I have the same problem (4.2.0-19-generic). After the release of the suspend -  black screen.