patjak/mba6x_bl #41: Closing laptop lid doesn't suspend with kernel 3.10.0-327.el7.x86_64 on RHEL 7.2

atodorov, 2015-11-21:
Yesterday I've upgraded to the latest RHEL 7.2 which includes kernel-3.10.0-327.el7.x86_64 . Just in case I've rebuilt the mba6x_bl driver and installed it. It seems to work but I have an issue with suspending. 

When laptop lid is closed the backlight stays on and drains my battery. This didn't happen with older kernels. OTOH if I point and click to Shutdown -> Suspend the backlight goes off and everything appears to work as expected. 

At this point I'm not sure if the latest Red Hat kernel has changed in a way that the drivers works partially (there was a change wrt the wl driver) or there's something else on the system not working properly.  Please let me know what logs I can collect to further debug the issue.

patjak, 2015-11-23:
A long shot but did you try
```echo XHC1 > /proc/acpi/wakeup```

Anything interesting in dmesg?

atodorov, 2015-11-24:
```
# cat /proc/acpi/wakeup 
Device	S-state	  Status   Sysfs node
PEG0	  S3	*disabled
EC	  S3	*disabled
HDEF	  S3	*disabled  pci:0000:00:1b.0
RP01	  S3	*disabled  pci:0000:00:1c.0
RP02	  S3	*disabled  pci:0000:00:1c.1
RP03	  S4	*disabled  pci:0000:00:1c.2
ARPT	  S4	*disabled  pci:0000:03:00.0
RP05	  S3	*disabled  pci:0000:00:1c.4
RP06	  S3	*disabled  pci:0000:00:1c.5
SPIT	  S3	*disabled  pnp:00:08
XHC1	  S3	*enabled   pci:0000:00:14.0
ADP1	  S3	*disabled
LID0	  S3	*enabled 

# echo XHC1 > /proc/acpi/wakeup

# cat /proc/acpi/wakeup 
Device	S-state	  Status   Sysfs node
PEG0	  S3	*disabled
EC	  S3	*disabled
HDEF	  S3	*disabled  pci:0000:00:1b.0
RP01	  S3	*disabled  pci:0000:00:1c.0
RP02	  S3	*disabled  pci:0000:00:1c.1
RP03	  S4	*disabled  pci:0000:00:1c.2
ARPT	  S4	*disabled  pci:0000:03:00.0
RP05	  S3	*disabled  pci:0000:00:1c.4
RP06	  S3	*disabled  pci:0000:00:1c.5
SPIT	  S3	*disabled  pnp:00:08
XHC1	  S3	*disabled  pci:0000:00:14.0
ADP1	  S3	*disabled
LID0	  S3	*enabled 
```

No effect after this change! What is see in the logs is the following:

After closing the lid, journalctl -f shows:
```
Nov 24 10:33:47 aero systemd-logind[974]: Lid closed.
```
After opening:
```
Nov 24 10:34:46 aero systemd-logind[974]: Lid opened.
```

When suspending from the menu:
```
Nov 24 10:35:18 aero NetworkManager[5176]: <info>  sleep requested (sleeping: no  enabled: yes)
Nov 24 10:35:18 aero NetworkManager[5176]: <info>  sleeping...
Nov 24 10:35:18 aero NetworkManager[5176]: <info>  (wlp3s0): device state change: activated -> unmanaged (reason 'sleeping') [100 10 37]
```

at this point the existing ssh connection freezes because the system is already suspended.  After reopening the laptop it resumes and I see various components of the system starting up. The one line which stands out as error is this:
```
Nov 24 10:35:59 aero kernel: mba6x_bl: Invalid response at reg: 0x1 (len: 24)
```

atodorov, 2015-11-27:
I've found the problem and it's not mba6x_bl. Fixed here:
https://bugzilla.redhat.com/show_bug.cgi?id=1284851

Closing.

closed 2015-11-27 by atodorov