patjak/mba6x_bl #42: Module no longer works with 4.2.0-19-generic, Ubuntu

proninyaroslav, 2015-12-19:
I have the same problem https://github.com/patjak/mba6x_bl/issues/40 (4.2.0-19-generic, /sys/class/backlight/mba6x_backlight not found). After the release of the suspend - black screen.
It helping
~~~
modporbe -r mba6x_bl
modprobe mba6x_bl
~~~
then appears /sys/class/backlight/mba6x_backlight

sbng, 2015-12-20:
I believe u need to compile and install from the source. I am running the same version of kernel without any issues. Follow the instruction in README.md.

proninyaroslav, 2015-12-20:
I follow the instructions, but i have this problem with version 4.2.0-19-generic.

napping, 2015-12-26:
I'm having the same problem as @proninyaroslav , and removing the driver and readding it fixes it for me as well.  Is there a solution for this?

proninyaroslav, 2016-01-16:
Now, if you unload a module system freezes hard.

proninyaroslav, 2016-01-18:
Remarked feature:
if not adjust the brightness of the buttons after loading notebook
```
$ ls /sys/class/backlight/
acpi_video0@ intel_backlight@
```
else if adjust the brightness of the buttons magically appear
```
$ ls /sys/class/backlight/
intel_backlight@ mba6x_backlight@
```

hpe009, 2016-01-20:
I am having the same problem, unloading mba6x_bl freezes the PC

patjak, 2016-01-20:
I'm having the same problem here. Will take a look.

hpe009, 2016-01-21:
@patjak Found anything interesting?

johnybx, 2016-01-22:
I am not sure if this information will help but I was able to install mba6x_bl on macbook air 7.1 Linux Mint 17.3 Rosa with kernel 4.4.0. After suspend brightness works same as before only f-keys doesn't work as they change values in /sys/class/backlight/intel_backlight/brightness not in /sys/class/backlight/mba6x_backlight/brightness but I can change brightness using echo.

hpe009, 2016-01-22:
@proninyaroslav Try:

 sudo gedit /lib/systemd/system-sleep/20_custom-ehci_hcd

Paste in:

#!/bin/sh
#inspired by http://art.ubuntuforums.org/showpost...0&postcount=19
#...and http://thecodecentral.com/2011/01/18...ot-working-bug
# tidied by tqzzaa :)
 VERSION=1.1
DEV_LIST=/tmp/usb-dev-list
DRIVERS_DIR=/sys/bus/pci/drivers
DRIVERS="ehci xhci" # ehci_hcd, xhci_hcd
HEX="[[:xdigit:]]"
MAX_BIND_ATTEMPTS=2
BIND_WAIT=0.1
 unbindDev() {
  echo -n > $DEV_LIST 2>/dev/null
   for driver in $DRIVERS; do
    DDIR=$DRIVERS_DIR/${driver}_hcd
    for dev in `ls $DDIR 2>/dev/null | egrep "^$HEX+:$HEX+:$HEX"`; do
      echo -n "$dev" > $DDIR/unbind
      echo "$driver $dev" >> $DEV_LIST
    done
   #for bus in $EHCI_BUSES; do
     echo -n $bus > /sys/bus/pci/drivers/ehci_hcd/unbind
 # done
   done
 }
 bindDev() {
  if [ -s $DEV_LIST ]; then
    while read driver dev; do
      DDIR=$DRIVERS_DIR/${driver}_hcd
      #for bus in $EHCI_BUSES; do
          echo -n $bus > /sys/bus/pci/drivers/ehci_hcd/bind
      #done
      while [ $((MAX_BIND_ATTEMPTS)) -gt 0 ]; do
          echo -n "$dev" > $DDIR/bind
          if [ ! -L "$DDIR/$dev" ]; then
            sleep $BIND_WAIT
          else
            break
          fi
          MAX_BIND_ATTEMPTS=$((MAX_BIND_ATTEMPTS-1))
      done
     done < $DEV_LIST
  fi
  rm $DEV_LIST 2>/dev/null
   chvt 1
  chvt 7
}
 EHCI_BUSES="0000:00:1a.0 0000:00:1d.0"
case "$1" in
    hibernate|suspend)
 unbindDev;;
     resume|thaw)
 bindDev;;
 esac

Save, and run:

 sudo chmod 755 /lib/systemd/system-sleep/20_custom-ehci_hcd

Reboot and try to suspend. May work

hpe009, 2016-01-22:
Try this:

sudo gedit /lib/systemd/system-sleep/20_xhc1_custom

Paste in:
```
#!/bin/sh
#inspired by http://art.ubuntuforums.org/showpost...0&postcount=19
#...and http://thecodecentral.com/2011/01/18...ot-working-bug
# tidied by tqzzaa :)
 VERSION=1.1
DEV_LIST=/tmp/usb-dev-list
DRIVERS_DIR=/sys/bus/pci/drivers
DRIVERS="ehci xhci" # ehci_hcd, xhci_hcd
HEX="[[:xdigit:]]"
MAX_BIND_ATTEMPTS=2
BIND_WAIT=0.1
 unbindDev() {
  echo -n > $DEV_LIST 2>/dev/null
   for driver in $DRIVERS; do
    DDIR=$DRIVERS_DIR/${driver}_hcd
    for dev in `ls $DDIR 2>/dev/null | egrep "^$HEX+:$HEX+:$HEX"`; do
      echo -n "$dev" > $DDIR/unbind
      echo "$driver $dev" >> $DEV_LIST
    done
   #for bus in $EHCI_BUSES; do
     echo -n $bus > /sys/bus/pci/drivers/ehci_hcd/unbind
 # done
   done
 }
 bindDev() {
  if [ -s $DEV_LIST ]; then
    while read driver dev; do
      DDIR=$DRIVERS_DIR/${driver}_hcd
      #for bus in $EHCI_BUSES; do
          echo -n $bus > /sys/bus/pci/drivers/ehci_hcd/bind
      #done
      while [ $((MAX_BIND_ATTEMPTS)) -gt 0 ]; do
          echo -n "$dev" > $DDIR/bind
          if [ ! -L "$DDIR/$dev" ]; then
            sleep $BIND_WAIT
          else
            break
          fi
          MAX_BIND_ATTEMPTS=$((MAX_BIND_ATTEMPTS-1))
      done
     done < $DEV_LIST
  fi
  rm $DEV_LIST 2>/dev/null
   chvt 1
  chvt 7
}
 EHCI_BUSES="0000:00:1a.0 0000:00:1d.0"
case "$1" in
    hibernate|suspend)
 unbindDev;;
     resume|thaw)
 bindDev;;
 esac
```

Save and enter this in terminal:

sudo chmod 755 /lib/systemd/system-sleep/20_xhc1_custom

Reboot and try to suspend

proninyaroslav, 2016-01-23:
@hpe009 
Thank you, but I can't verify, since switched to Fedora.

hpe009, 2016-02-10:
@patjak Systemd might be the issue, try switching to upstart

patjak, 2016-02-12:
Pushed some changes. Removing the driver should not hang the machine anymore.

patjak, 2016-04-09:
Try the latest code and open new issue of problems still exists.

closed 2016-04-09 by patjak