patjak/mba6x_bl #43: Minimum brightness, can't work around it

hobarrera, 2016-02-06:
I've installed the driver on linux-4.4.1. I then rebooted, and the system sets brightness to a minimum during boot.

After several reboots, I managed to regain access to the machine by sshing into the macbook. Brightness could not be altered (attempting to alter it had no result).

Unloading the module and then changing brightness results in a kernel panic. I had to uninstall it and reboot to recover my machine.

hobarrera, 2016-02-15:
There have been more reports of the same issues on linux>=4.4:
https://aur.archlinux.org/packages/mba6x_bl-dkms/

aminb, 2016-02-16:
Yeah, same issue here.

mojoman, 2016-02-28:
Same issue here with 4.4.0 and latest source from git, black screen on X startup. However, if I (blindly) type my login/password for the desktop-manager, backlight is on for the session and works find on suspend/resume. Weird :)

aminb, 2016-02-28:
With the issue present 4.3.6, I've written this systemd unit for myself as a temporary workaround:

````
# copy to /etc/systemd/system/ (at least on Arch Linux)

[Unit]
Description=Remove and reload mba6x to workaround no brightness bug

[Service]
ExecStart=/bin/sh -c "modprobe -r mba6x_bl && modprobe mba6x_bl"
Type=oneshot

[Install]
WantedBy=multi-user.target
```

Save it to a file and copy it to `/etc/systemd/system/`, then enable it with `sudo systemctl enable FILENAME`.

mojoman, 2016-02-28:
Yes, that helps with 4.4.0 too. Thanks!

hobarrera, 2016-02-29:
@aminb So removing the module after boot has completed and re-inserting it fixes the issue?

aminb, 2016-02-29:
@mojoman Glad to hear it helped!

@hobarrera Yes, as a temporary workaround. I can confirm that it works on 4.3.6, and apparently it works on 4.4.0 too as confirmed by @mojoman.

I saw this workaround mentioned in #42 and I thought I'd give it a shot, and it worked so I wrote a systemd unit file for myself.

Hopefully @patjak can come up with a proper fix for this.

hobarrera, 2016-03-30:
The above systemd file alone was not enough for me. Both `xbacklight` and [`light`](https://github.com/haikarainen/light) seemed to be using the wrong controller by default (`intel_backlight`?). Expliclty forcing the latter to use the `mba6x_backlight` controller fixed the issue for me - once I'd added the above systemd file.

Is it somehow possible to disable that alternate driver entirely?

Poohblah, 2016-03-31:
@patjak, any status update on this?

patjak, 2016-04-03:
Don't know what's going on here. I cannot reproduce it myself. Everything works fine with every kernel version on my MBA 6,2.

Poohblah, 2016-04-04:
Hmm, that certainly puts a damper on debugging!  I believe my model is an 11" 7,1.

rmanne, 2016-04-04:
I can confirm, @aminb 's fix helps for 4.4.5-1 as well.

@patjak I have an MBA 6,2 as well. When you reboot, do you notice how the screen is just completely black? Of course, one easy way to bring back the screen is to make it suspend (I have my power button setup to make it suspend) and then resume. But the script makes it so that I don't have to do this. I'm not sure at all why there is an issue at all, but I can confirm that this issue does exist on the MBA 6,2, perhaps some settings you use make this a non-issue. And additionally, it's almost impossible to tell from the dmesg, I didn't even notice that this driver was causing the issue until that PKGBUILD got updated and I recently reinstalled arch.

@hobarrera Yes I believe the name of the module you're looking for is "apple_bl". So blacklisting it should disable that alternate driver entirely.

aminb, 2016-04-04:
@patjak One way that you might be able to reproduce the issue would be to do a clean install of your OS. I noticed this issue when I upgraded to 4.4, but reverting back to 4.3 resolved the issue (I wasn't getting the black screen anymore).

Then, something came up and I had to do a clean install of Arch, then even with versions <4.4 I face this problem. The problem also happens with with the linux-ck kernel.

patjak, 2016-04-06:
It could be that mba6x_bl is getting loaded before i915 . Perhaps it's just time to give up on solving this correctly and instead let mba6x_bl periodically (like every 3 secs) reset the backlight level. That wouldn't be optimal for battery life but would probably fix the issues people are seeing.

rmanne, 2016-04-06:
@patjak Hmm, if what you say is true and the error is in that mba6x_bl is getting loaded before i915, then wouldn't a valid fix to explicitly list the dependency? That i915 should load before mba6x_bl?

patjak, 2016-04-09:
@rmanne perhaps, if we can do that? I caved in to the pressure and added reprogramming of the entire chip every 3 seconds. If this still doesn't work for people, the problem must be somewhere else.

Poohblah, 2016-06-09:
I haven't updated my kernel in 5 months to avoid breaking the backlight.  Has anybody had any luck with any of the changes that @patjak made? I need to get real, actual work done on this laptop at the moment so I haven't had the opportunity to test :stuck_out_tongue_winking_eye:

rmanne, 2016-06-12:
wat, as we stated earlier, you can always just try aminb's fix for it. I'm on 4.6.2 and I have no problems with it right now. (I didn't update to the latest version yet though, it's 1 before the patch mentioned in this thread I think.)

brahm92, 2016-06-18:
I want to test out the earlier driver (one before the workarounds were implemented) and see if the issue persists on 4.6.2-1 kernel on arch. Can someone give me a few pointers on how I can do this? Currently i have the version from aur installed.