patjak/mba6x_bl #44: Screen gets black after ~20 suspend to ram (acpitool -s)

delanoe, 2016-03-21:
The screen gets black, need to suspend to ram again and at startup, it is fine untile some minutes. I have to restart and issue appears again after ~20 suspend to ram. 

Maybe it is not related to mba6x_bl but it is installed on the system (so I guess, there is a link, sorry if not)

Here is the dmesg when screen gets black:

Mar 21 12:28:59 linux kernel: [70868.369112] ------------[ cut here ]------------
Mar 21 12:28:59 linux kernel: [70868.369130] WARNING: CPU: 3 PID: 23245 at /build/linux-P15SNz/linux-3.16.3/drivers/gpu/drm/i915/intel_pm.c:6001 intel_display_power_put+0x111/0x150 [i915]()
Mar 21 12:28:59 linux kernel: [70868.369149] Modules linked in: asix usbnet libphy mii usb_storage xt_conntrack ipt_MASQUERADE iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype iptable_filter ip_tables x_tables nf_nat nf_conntrack bridge stp llc aufs(C) pci_stub vboxpci(O) vboxnetadp(O) binfmt_misc vboxnetflt(O) bnep vboxdrv(O) nfsd auth_rpcgss oid_registry nfs_acl nfs lockd fscache sunrpc nls_utf8 nls_cp437 vfat fat snd_hda_codec_hdmi snd_hda_codec_cirrus snd_hda_codec_generic iTCO_wdt iTCO_vendor_support ecb btusb bluetooth 6lowpan_iphc joydev mba6x_bl(O) bcm5974 applesmc input_polldev wl(PO) evdev x86_pkg_temp_thermal intel_powerclamp efi_pstore intel_rapl coretemp kvm_intel kvm cfg80211 pcspkr snd_hda_intel snd_hda_controller efivars rfkill snd_hda_codec lpc_ich i2c_i801 mfd_core snd_hwdep snd_pcm_oss snd_mixer_oss snd_pcm snd_seq_midi snd_seq_midi_event snd_rawmidi snd_seq i915 snd_seq_device sbs snd_timer sbshc drm_kms_helper ac snd drm video apple_bl battery i2c_algo_bit i2c_core soundcore button mei_me shpchp mei processor thermal_sys loop fuse parport_pc ppdev lp parport autofs4 ext4 crc16 mbcache jbd2 algif_skcipher af_alg hid_generic hid_apple usbhid hid dm_crypt dm_mod sg sd_mod crc_t10dif crct10dif_generic crct10dif_pclmul crct10dif_common crc32_pclmul crc32c_intel ghash_clmulni_intel aesni_intel aes_x86_64 lrw gf128mul glue_helper ablk_helper cryptd ahci libahci libata xhci_hcd scsi_mod usbcore usb_common
Mar 21 12:28:59 linux kernel: [70868.369170] CPU: 3 PID: 23245 Comm: kworker/u16:24 Tainted: P        WC O  3.16-2-amd64 #1 Debian 3.16.3-2
Mar 21 12:28:59 linux kernel: [70868.369171] Hardware name: Apple Inc. MacBookAir6,1/Mac-35C1E88140C3E6CF, BIOS MBA61.88Z.0099.B09.1402071141 02/07/2014
Mar 21 12:28:59 linux kernel: [70868.369174] Workqueue: events_unbound async_run_entry_fn
Mar 21 12:28:59 linux kernel: [70868.369175]  0000000000000009 ffffffff81506188 0000000000000000 ffffffff81065707
Mar 21 12:28:59 linux kernel: [70868.369176]  0000000000000000 ffffffffa04417a0 ffff880036980000 0000000000800000
Mar 21 12:28:59 linux kernel: [70868.369177]  ffff880036988520 ffffffffa0399151 ffff880036980000 0000000000000000
Mar 21 12:28:59 linux kernel: [70868.369178] Call Trace:
Mar 21 12:28:59 linux kernel: [70868.369182]  [<ffffffff81506188>] ? dump_stack+0x41/0x51
Mar 21 12:28:59 linux kernel: [70868.369184]  [<ffffffff81065707>] ? warn_slowpath_common+0x77/0x90
Mar 21 12:28:59 linux kernel: [70868.369190]  [<ffffffffa0399151>] ? intel_display_power_put+0x111/0x150 [i915]
Mar 21 12:28:59 linux kernel: [70868.369198]  [<ffffffffa03defb4>] ? intel_display_set_init_power+0x24/0x40 [i915]
Mar 21 12:28:59 linux kernel: [70868.369205]  [<ffffffffa03dfc8e>] ? __intel_set_mode+0x5ce/0x1560 [i915]
Mar 21 12:28:59 linux kernel: [70868.369212]  [<ffffffffa03ca514>] ? gen6_read32+0x64/0x120 [i915]
Mar 21 12:28:59 linux kernel: [70868.369214]  [<ffffffff8150a9ee>] ? mutex_lock+0xe/0x2a
Mar 21 12:28:59 linux kernel: [70868.369221]  [<ffffffffa03e4fb4>] ? intel_modeset_setup_hw_state+0x824/0xd30 [i915]
Mar 21 12:28:59 linux kernel: [70868.369225]  [<ffffffffa0344b5a>] ? drm_modeset_lock+0x2a/0xd0 [drm]
Mar 21 12:28:59 linux kernel: [70868.369229]  [<ffffffffa0344c41>] ? drm_modeset_lock_all_crtcs+0x41/0x60 [drm]
Mar 21 12:28:59 linux kernel: [70868.369231]  [<ffffffff812e0420>] ? pci_pm_thaw+0x90/0x90
Mar 21 12:28:59 linux kernel: [70868.369235]  [<ffffffffa038c37e>] ? __i915_drm_thaw+0x11e/0x1a0 [i915]
Mar 21 12:28:59 linux kernel: [70868.369239]  [<ffffffffa038cbd4>] ? i915_resume+0x24/0x40 [i915]
Mar 21 12:28:59 linux kernel: [70868.369241]  [<ffffffff813a8a96>] ? dpm_run_callback+0x46/0xf0
Mar 21 12:28:59 linux kernel: [70868.369242]  [<ffffffff813a8ec6>] ? device_resume+0x86/0x1a0
Mar 21 12:28:59 linux kernel: [70868.369243]  [<ffffffff813a8ff9>] ? async_resume+0x19/0x40
Mar 21 12:28:59 linux kernel: [70868.369244]  [<ffffffff8108b842>] ? async_run_entry_fn+0x32/0x120
Mar 21 12:28:59 linux kernel: [70868.369246]  [<ffffffff8107f4a2>] ? process_one_work+0x172/0x420
Mar 21 12:28:59 linux kernel: [70868.369248]  [<ffffffff8107fb35>] ? worker_thread+0x115/0x520
Mar 21 12:28:59 linux kernel: [70868.369249]  [<ffffffff8107fa20>] ? rescuer_thread+0x2d0/0x2d0
Mar 21 12:28:59 linux kernel: [70868.369251]  [<ffffffff81085dcd>] ? kthread+0xbd/0xe0
Mar 21 12:28:59 linux kernel: [70868.369252]  [<ffffffff81085d10>] ? kthread_create_on_node+0x180/0x180
Mar 21 12:28:59 linux kernel: [70868.369253]  [<ffffffff8150c1bc>] ? ret_from_fork+0x7c/0xb0
Mar 21 12:28:59 linux kernel: [70868.369254]  [<ffffffff81085d10>] ? kthread_create_on_node+0x180/0x180
Mar 21 12:28:59 linux kernel: [70868.369255] ---[ end trace bf954e2fb26212d1 ]---
Mar 21 12:28:59 linux kernel: [70868.672710] ata1: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
Mar 21 12:28:59 linux kernel: [70868.673074] ata1.00: unexpected _GTF length (8)
Mar 21 12:28:59 linux kernel: [70868.673794] ata1.00: unexpected _GTF length (8)
Mar 21 12:28:59 linux kernel: [70868.673985] ata1.00: configured for UDMA/133
Mar 21 12:28:59 linux kernel: [70868.849280] PM: resume of devices complete after 520.492 msecs
Mar 21 12:28:59 linux kernel: [70868.849391] PM: Finishing wakeup.
Mar 21 12:28:59 linux kernel: [70868.849392] Restarting tasks ... done.
Mar 21 12:28:59 linux kernel: [70868.854211] cfg80211: World regulatory domain updated:
Mar 21 12:28:59 linux kernel: [70868.854214] cfg80211:  DFS Master region: unset
Mar 21 12:28:59 linux kernel: [70868.854216] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
Mar 21 12:28:59 linux kernel: [70868.854219] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
Mar 21 12:28:59 linux kernel: [70868.854221] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
Mar 21 12:28:59 linux kernel: [70868.854223] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
Mar 21 12:28:59 linux kernel: [70868.854225] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
Mar 21 12:28:59 linux kernel: [70868.854227] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
Mar 21 12:28:59 linux kernel: [70868.854229] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
Mar 21 12:28:59 linux kernel: [70868.854231] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
Mar 21 12:28:59 linux kernel: [70868.854232] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
Mar 21 12:29:00 linux kernel: [70870.008020] [drm] Enabling RC6 states: RC6 on, RC6p off, RC6pp off
Mar 21 12:29:00 linux kernel: [70870.121144] asix 1-2:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1
Mar 21 12:29:02 linux kernel: [70871.583253] ------------[ cut here ]------------
Mar 21 12:29:02 linux kernel: [70871.583288] WARNING: CPU: 0 PID: 15054 at /build/linux-P15SNz/linux-3.16.3/drivers/gpu/drm/i915/intel_pm.c:6001 intel_display_power_put+0x111/0x150 [i915]()
Mar 21 12:29:02 linux kernel: [70871.583290] Modules linked in: asix usbnet libphy mii usb_storage xt_conntrack ipt_MASQUERADE iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype iptable_filter ip_tables x_tables nf_nat nf_conntrack bridge stp llc aufs(C) pci_stub vboxpci(O) vboxnetadp(O) binfmt_misc vboxnetflt(O) bnep vboxdrv(O) nfsd auth_rpcgss oid_registry nfs_acl nfs lockd fscache sunrpc nls_utf8 nls_cp437 vfat fat snd_hda_codec_hdmi snd_hda_codec_cirrus snd_hda_codec_generic iTCO_wdt iTCO_vendor_support ecb btusb bluetooth 6lowpan_iphc joydev mba6x_bl(O) bcm5974 applesmc input_polldev wl(PO) evdev x86_pkg_temp_thermal intel_powerclamp efi_pstore intel_rapl coretemp kvm_intel kvm cfg80211 pcspkr snd_hda_intel snd_hda_controller efivars rfkill snd_hda_codec lpc_ich i2c_i801 mfd_core sn [... cut ...]
Mar 21 12:32:57 linux kernel: [71107.035376]  [<ffffffff8107fa20>] ? rescuer_thread+0x2d0/0x2d0
Mar 21 12:32:57 linux kernel: [71107.035381]  [<ffffffff81085dcd>] ? kthread+0xbd/0xe0
Mar 21 12:32:57 linux kernel: [71107.035386]  [<ffffffff81085d10>] ? kthread_create_on_node+0x180/0x180
Mar 21 12:32:57 linux kernel: [71107.035390]  [<ffffffff8150c1bc>] ? ret_from_fork+0x7c/0xb0
Mar 21 12:32:57 linux kernel: [71107.035395]  [<ffffffff81085d10>] ? kthread_create_on_node+0x180/0x180
Mar 21 12:32:57 linux kernel: [71107.035398] ---[ end trace bf954e2fb26212d6 ]--

patjak, 2016-03-26:
Hi, the problem is in the i915 driver and not in mba6x_bl. Looks like you're running a pretty old kernel (3.16?). You could try upgrading and see if that helps.

delanoe, 2016-03-27:
Hello,

2016/03/25 19:24, Patrik Jakobsson:
> Hi, the problem is in the i915 driver and not in mba6x_bl. Looks like you're
> running a pretty old kernel (3.16?). You could try upgrading and see if that
> helps.

Thanks for your reply.

I have installed 4.4.0-1-amd64 kernel.

So I will see if it happens again after 20 times of acpitool -s.

Cheers,

-- 
Alexandre Delanoë

patjak, 2016-04-06:
Hi, any update on this? Can we close the issue?

closed 2016-04-09 by patjak

delanoe, 2016-04-19:
2016/04/05 23:08, Patrik Jakobsson:
> Hi, any update on this? Can we close the issue?

Yes, I think you can. Sorry for the delay (I am still testing in fact,
but I think it is okay).

-- 
Alexandre Delanoë