patjak/mba6x_bl #46: Error compyling

kuramaluka, 2016-04-23:
I'm using ubuntu 16.04 on my macbook air, when I try to compile the package it turn ...

`luca@Freedom:~/Scaricati/mba6x_bl$ sudo make install
[sudo] password di luca: 
make -C /lib/modules/4.4.0-21-generic/build M=/home/luca/Scaricati/mba6x_bl modules_install
make[1]: ingresso nella directory "/usr/src/linux-headers-4.4.0-21-generic"
  INSTALL /home/luca/Scaricati/mba6x_bl/mba6x_bl.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-21-generic
make[1]: uscita dalla directory "/usr/src/linux-headers-4.4.0-21-generic"`

Any idea on what is the problem? Thank you!

miekg, 2016-04-23:
The building worked. This happens when you install the .ko file. It should
just work, the error is harmless.

sami10007, 2016-06-08:
I get the same error. A thread about it here http://askubuntu.com/q/784325/25388 Any verifications about the issue?

sami10007, 2016-06-08:
I can confirm miekg's arguments. I get the same errors. The error is harmless. The backlight control seems to work know.

duryjames, 2016-11-03:
Sorry to comment on a closed error

Can you explain this a bit more? 

I am getting this error on my macbook air ubuntu 16.04 and am not sure what to do after 

    $ sudo make install 
    make -C /lib/modules/4.4.0-45-generic/build M=/home/user/Downloads/mba6x_bl modules_install
    make[1]: Entering directory '/usr/src/linux-headers-4.4.0-45-generic'
    INSTALL /home/user/Downloads/mba6x_bl/mba6x_bl.ko
    At main.c:222:
    - SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
    - SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
    sign-file: certs/signing_key.pem: No such file or directory
      DEPMOD  4.4.0-45-generic
    make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-45-generic'

I don't believe its installed as running:

    sudo modprobe mba6x_bl

causes:

    modprobe: FATAL: Module mba6x_bl not found in directory /lib/modules/4.4.0-45-generic

duryjames, 2016-11-03:
Ok ignore my last comment

Followed this

    https://github.com/patjak/mba6x_bl/issues/54

And got everything working