patjak/mba6x_bl #47: Set backlight level

engstk, 2016-05-02:
I build the driver on MacBook Air 6,1 early 2014 with Ubuntu 16.04 and Linux 4.5.2, working good.
After shutdown or reboot, the driver always load a default level of 50~60%, not saving the last used value like in stock driver.
Is it possible the change this behavior or is it a known issue?

patjak, 2016-05-02:
I actually decided not to set full brightness on boot so I could see when the driver kicks in. It can be changed to 100% if people think it's better. The issue have been raised before.

The OSX driver can remember the backlight setting and I think this is stored in nvram somewhere. I rather not do that in our driver and I think distros should take care of storing/restoring the brightness from userspace.

engstk, 2016-05-03:
Yes exactly, full brightness it's not a good idea.
I'm not talking about the OSX driver but rather the default backlight ACPI driver without this mod.
It saves the last used brightness in Ubuntu 16.04, it just doesn't saves it if using this (it forces the 50~60% like said)

patjak, 2016-09-28:
Closing this for now. We can reopen if we find some good solution to this that can be implemented in the driver. It is possible that the default value gets sent to the wrong backlight device and never reaching mba6x_bl. We're racing a bit with i915 over the backlight control and not sure how we should handle this. I've been talking to the i915 backlight devs and they are a bit hesitant (with good reasons) to add special handling of specific hardware configs like the MacBook. Possibly we could add a notifier to backlight devices so that every device that receives a backlight change can notify other backlight devices.

closed 2016-09-28 by patjak