patjak/mba6x_bl #49: stops working after modifying /proc/acpi/wakeup

dsvid, 2016-07-07:
Thanks for making this, it's very useful!

I have found an issue, however, when I went to change my wakeup options to disable XHC1 in my /proc/acpi/wakeup settings (as my Macbook was waking frequently with the lid closed).

Unfortunately, when I disabled this, while my issue with waking with the lid closed was resolved, the backlight driver fix stopped working, and my backlight went back to turning off upon resume (the fix was previously working).

patjak, 2016-07-07:
Hi,

wakeup settings shouldn't affect mba6x_bl. Are you sure this is what is causing it? Anything odd in dmesg when this happens? Can you still change the backlight through /sys/class/backlight/mba6x_bl/brightness ?

dsvid, 2016-07-07:
Hey, thanks for the message! The reason I assumed that it was due to the
wakeup setting change is that it was working fine right up until I changed
the wakeup setting, and I didn't change anything else.

I'm not at my computer right now but I will check those things that you
suggested tomorrow.

Thanks for taking the time to try to help with this.

dsvid, 2016-07-10:
hi, it looks like for some reason your driver got removed from the kernel.
I don't know what caused this, but it seemed to just be a coincidence that
it happened around the same time as my changing the wakeup options.

I have re-loaded your driver and it tentatively seems to be working now
even after changing the wakeup settings.

Thanks for the help!

patjak, 2016-07-10:
That would explain it :)
Great that it's working now.

closed 2016-07-10 by patjak

dsvid, 2016-07-10:
Thanks for the help! Really useful driver btw :)