patjak/mba6x_bl #50: mba6x_bl has no effect

half-potato, 2016-07-24:
Nothing looks suspicious from the logs, but the backlight still only goes from 100% - 0% after suspending my Macbook Air 6,2 (Early 2014). Using Ubuntu 14.01 with the 4.2.0-27-generic kernel.

-> make
make -C /lib/modules/4.2.0-27-generic/build M=/home/amai/mba6x_bl modules
make[1]: Entering directory `/usr/src/linux-headers-4.2.0-27-generic'
  Building modules, stage 2.
  MODPOST 1 modules
make[1]: Leaving directory `/usr/src/linux-headers-4.2.0-27-generic'
-> sudo make install                
make -C /lib/modules/4.2.0-27-generic/build M=/home/amai/mba6x_bl modules_install
make[1]: Entering directory `/usr/src/linux-headers-4.2.0-27-generic'
  INSTALL /home/amai/mba6x_bl/mba6x_bl.ko
  DEPMOD  4.2.0-27-generic
make[1]: Leaving directory `/usr/src/linux-headers-4.2.0-27-generic'
-> sudo depmod -a
-> sudo modprobe mba6x_bl
-> dmesg | tail -20
[25808.464370] ata1.00: status: { DRDY }
[25808.464374] ata1: hard resetting link
[25809.188368] ata1: SATA link up 1.5 Gbps (SStatus 113 SControl 310)
[25809.188891] ata1.00: unexpected _GTF length (8)
[25809.189504] ata1.00: unexpected _GTF length (8)
[25809.189582] ata1.00: configured for UDMA/33
[25809.189649] ata1: EH complete
[26298.072173] ata1.00: exception Emask 0x10 SAct 0x1000000 SErr 0x4040000 action 0xe frozen
[26298.072176] ata1.00: irq_stat 0x80000040, connection status changed
[26298.072178] ata1: SError: { CommWake DevExch }
[26298.072181] ata1.00: failed command: WRITE FPDMA QUEUED
[26298.072184] ata1.00: cmd 61/18:c0:28:c3:ff/00:00:0b:00:00/40 tag 24 ncq 12288 out
[26298.072184]          res 40/00:bc:80:7e:fe/00:00:0b:00:00/40 Emask 0x10 (ATA bus error)
[26298.072186] ata1.00: status: { DRDY }
[26298.072188] ata1: hard resetting link
[26298.796114] ata1: SATA link up 1.5 Gbps (SStatus 113 SControl 310)
[26298.796591] ata1.00: unexpected _GTF length (8)
[26298.797132] ata1.00: unexpected _GTF length (8)
[26298.797220] ata1.00: configured for UDMA/33
[26298.797270] ata1: EH complete

half-potato, 2016-07-24:
Solution is to restart it
sudo modprobe -r mba6x_bl
sudo modprobe mba6x_bl

patjak, 2016-09-28:
Seems your system is using the wrong backlight device and reloading it makes it the default again. Not sure what is causing this unfortunately.