patjak/mba6x_bl #52: make install error on ubuntu 16.04

awjrichards, 2016-08-18:
On Ubuntu 16.04 with my MacBook Air 6,2 I get the following error on make install:

```
# make install
make -C /lib/modules/4.4.0-31-generic/build M=/home/awjrichards/src/mba6x_bl modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-31-generic'
  INSTALL /home/awjrichards/src/mba6x_bl/mba6x_bl.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-31-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-31-generic'
```

awjrichards, 2016-08-18:
I tried again using `checkinstall` rather than than `make install` and it worked just fine.

patjak, 2016-09-28:
Thanks for the info. Closing the issue since this seems resolved. The issue itself should be that you're missing libssl.

closed 2016-09-28 by patjak