patjak/mba6x_bl #53: Having some trouble after reboot

mortenskoett, 2016-09-25:
Hi.

Macbook Air 6,2. 2013
Xubuntu.

I have had this wonderful driver working for a couple of days. Great work. But then...

Problem:
I tried to minimize power consumption with TLP and laptop-mode-tools, then the system suddenly didn't want to stay suspended and kind of woke up just after, sadly with the not backlighted screen.

I uninstalled alle the power config tools and reinstalled mba6x_bl to find it works but ONLY until I reboot.

I then added   mba6x_bl to /etc/modules  hoping for the system to reinitialise every time, but now I'm back to the point where it won't stay suspended.

Can anybody give me some advice? It's horrible to being able to suspend the darn laptop.

Thanks!

M

patjak, 2016-09-28:
The sudden wakeups could be related to USB wakeup events. Have you tried typing (as root):
```echo XHCI > /proc/acpi/wakeup```

dispicio, 2016-10-14:
I have had a similar problem (with Powertop and the tunable "Autosuspend for USB device xHCI Host Controller [usb2]")

After
`echo XHCI > /proc/acpi/wakeup`
the problem was resolved.

Thank you :)