patjak/mba6x_bl #54: Install error on 16.04

betolink, 2016-09-27:
I tried to install the driver on Ubuntu 16.04 (kernel 4.4.0-38-generic), after compiling I get 

```
make -C /lib/modules/4.4.0-38-generic/build M=/home/beto/hacks/mba6x_bl modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.4.0-38-generic'
  INSTALL /home/beto/hacks/mba6x_bl/mba6x_bl.ko
At main.c:222:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.4.0-38-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-38-generic'
```

and when I tried to load the module I get:

```
modprobe: FATAL: Module mba6x_bl not found in directory /lib/modules/4.4.0-38-generic
```

Is the driver supported for 16.04?

patjak, 2016-09-28:
Try installing libssl, that should help.

patjak, 2016-09-28:
Also, see issue #52

patjak, 2016-09-28:
...and issue #46

closed 2016-09-28 by patjak

betolink, 2016-09-28:
I have ssl installed, looks like this error is related to signed modules on the linux kernel. i.e. [Module signature verification issues on Ubuntu 16.04 LTS](https://github.com/darlinghq/darling/issues/152) and [Ubuntu 16.04, make install leads to ssl error](https://github.com/free5lot/hid-apple-patched/issues/23)

patjak, 2016-09-29:
Did you install the development version of the package (libssl-dev on Ubuntu). Yes, you need it to be able to sign modules. Either way, this is not a fatal error so you should still be able to use the driver.

betolink, 2016-09-29:
Yes I have libssl-dev installed and when I tried to load the module I get the error I put on my first comment: 

`modprobe: FATAL: Module mba6x_bl not found in directory /lib/modules/4.4.0-38-generic`

I finally got it working by following @luveti steps on [#52](https://github.com/darlinghq/darling/issues/152), 
* `sudo update-initramfs -u -k all`
* `sudo insmod /lib/modules/linux-image-$(uname -r)/extra/mba6x_bl.ko`

tpcorrea, 2017-01-04:
Hi all,

sorry to comment a closed issue, but I have exactly this problem, except that when I use insmod, the screen dims, but when I re-start Ubuntu everything is as before, i.e., I can't control the backlight...

Could anyone help? Cheers.

emily-xx, 2017-08-14:
@tpcorrea The way I got things to work all the time was by adding the insmod portion of things to /etc/rc.local

```
#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

echo 0 > /sys/module/hid_apple/parameters/iso_layout
insmod /lib/modules/$(uname -r)/extra/mba6x_bl.ko

exit 0
```

You need to do the update-initramfs -u -k all after installing the module for a new kernel, then do a reboot and that line will ensure that the module is loaded on each boot!

I use KDE Neon which is built atop an Ubuntu 16.04 base, so that command should work fine for you if using Ubuntu 16.04. If not, check out how your module directories are named in /lib/modules and adjust accordingly.

The echo 0 part of things fixes the issue where the tilda key produces ", as seen here: https://askubuntu.com/questions/530325/tilde-key-on-mac-air-with-ubuntu