patjak/mba6x_bl #55: building problems

damdamce, 2016-10-02:
hi,
i have building problems after installing mba6x_bl to linux mint using the ubuntu deb package. it worked before, but apparently i changed something to the system and it stopped.

    madams-mint-on-MacBookAir madam # dkms install mba6xbl/1.0.0

    Kernel preparation unnecessary for this kernel.  Skipping...

    Building module:
    cleaning build area....
    make KERNELRELEASE=4.4.0-21-generic all KVERSION=4.4.0-21-generic....(bad exit status: 2)
    Error! Bad return status for module build on kernel: 4.4.0-21-generic (x86_64)
    Consult /var/lib/dkms/mba6xbl/1.0.0/build/make.log for more information.
    madams-mint-on-MacBookAir madam # cat /var/lib/dkms/mba6xbl/1.0.0/build/make.log
    DKMS make.log for mba6xbl-1.0.0 for kernel 4.4.0-21-generic (x86_64)
    Sun Oct  2 19:14:54 EEST 2016
    make -C /lib/modules/4.4.0-21-generic/build M=/var/lib/dkms/mba6xbl/1.0.0/build modules
    make[1]: Entering directory '/usr/src/linux-headers-4.4.0-21-generic'
    CC [M]  /var/lib/dkms/mba6xbl/1.0.0/build/mba6x_bl.o
    /var/lib/dkms/mba6xbl/1.0.0/build/mba6x_bl.c: In function ‘platform_probe’:
    /var/lib/dkms/mba6xbl/1.0.0/build/mba6x_bl.c:336:2: error: implicit declaration of function ‘acpi_video_dmi_promote_vendor’ [-Werror=implicit-function-declaration]
    acpi_video_dmi_promote_vendor();
    ^
    cc1: some warnings being treated as errors
    scripts/Makefile.build:264: recipe for target '/var/lib/dkms/mba6xbl/1.0.0/build/mba6x_bl.o' failed
    make[2]: *** [/var/lib/dkms/mba6xbl/1.0.0/build/mba6x_bl.o] Error 1
    Makefile:1396: recipe for target '_module_/var/lib/dkms/mba6xbl/1.0.0/build' failed
    make[1]: *** [_module_/var/lib/dkms/mba6xbl/1.0.0/build] Error 2
    make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-21-generic'
    Makefile:8: recipe for target 'all' failed
    make: *** [all] Error 2

i'm not sure this is actually an issue with the code, but maybe you have an idea on what configuration is causing it. i am following documentation in https://help.ubuntu.com/community/MacBookAir6-2/Trusty to improve energy usage and make things work.

damdamce, 2016-10-02:
i just found https://github.com/patjak/mba6x_bl/issues/45

so it seems like the problem is just in the deb package, but why did it work before?

damdamce, 2016-10-02:
because the deb package is not up to date.
this commit https://github.com/patjak/mba6x_bl/commit/0b095b4c75666a7dab17d7c919f59b3b8426afc7 is necessary for my kernel. i'm just surprised it worked in the first place. i made a kernel update, but both are 4.4.0. i don't know much about it, maybe they patched it so the required function was there.

anyway, please read this bugreport as "could you update the packages?" or at least a warning that they don't work for newer kernels? thanks :)

damdamce, 2016-10-02:
i built from trunk and it works on my system.
thanks for your work :)

[mba6xbl-dkms_1.1.0_all.deb.zip](https://github.com/patjak/mba6x_bl/files/505288/mba6xbl-dkms_1.1.0_all.deb.zip)

ivanperez-keera, 2017-06-02:
Would it be possible to create a deb source package with travis and upload it to a PPA, or connect the repo with launchpad so that it's built automatically? That would keep it always up to date.