patjak/mba6x_bl #57: Include in mainline kernel

atodorov, 2016-10-27:
@patjak what is your intention for this driver and if going to upstream it what is the current status? 

I see in other issues people trying to build for the 4.x kernel series and reporting some problems. Maybe you can update README or the wiki with some short info as well.

Thanks.

patjak, 2016-10-27:
The mainline branch works fine with current kernels so use that unless it doesn't compile for your particular kernel. I'm trying to get suspend/resume working now and then I will send it upstream.

atodorov, 2016-10-27:
Thanks. Feel free to close this if you like but please post a link when you send it  upstream.

patjak, 2016-10-27:
Oh, sorry wrong project. I was thinking about bcwc_pcie :)
I don't have plans to upstream this right now since it's still seems broken on some systems. It's a workaround that works for me but not sure if upstreaming it would break more than it fixes.

alex-robbins, 2017-03-06:
I noticed a couple months ago that I'm now able to use my 2014 MBA's backlight properly, after resuming from suspend, even without mba6x_bl installed.  I'm using the standard Debian Testing kernel (currently 4.9).

Could it be that this has been fixed upstream?

atodorov, 2017-03-06:
@alex-robbins I don't think this driver has been merged upstream. However I've also noticed improvement in the backlight driver. On my RHEL 7 system, after upgrade I saw an updated intel_backlight driver which appears to work. I have removed mba6x_bl-kmod from my system but it still remains in initrd. When I unload it backlight seems to work fine but not without mba6x_bl totally gone.

patjak, 2017-03-06:
I just tried 4.8, 4.9 and 4.10 without mba6x_bl. 4.8 still has issues but with 4.9 and 4.10 the issue seems to be gone! I'll take a look and see if I can find which commit fixes this and if it can be backported. \o/

patjak, 2017-03-06:
I haven't tested it yet but I'm pretty sure these are the commits that fixes the issue:

```
commit 16e1203db8ab740c912ea62761fdf27d7811b886
Author: Jani Nikula <jani.nikula@intel.com>
Date:   Mon Sep 19 13:35:25 2016 +0300

    drm/i915/backlight: setup and cache pwm alternate increment value
```

```
commit 915b417946030e1f365c63728875cffd8db8e880
Author: Shawn Lee <shawn.c.lee@intel.com>
Date:   Mon Sep 19 13:35:26 2016 +0300

    drm/i915/backlight: setup backlight pwm alternate increment on backlight enable
```