patjak/mba6x_bl #58: Brightness not changing on latest xf86-video-intel

aminb, 2016-10-31:
On Arch, `xf86-video-intel` was bumped to latest master 2 days ago, and it breaks `mba6x_bl` such that increasing or decreasing the brightness has no effect. The value reported by `xbacklight` does change, but there's no actual change in how bright the screen is.

Last known good commit: https://cgit.freedesktop.org/xorg/driver/xf86-video-intel/commit/?id=eb01cc549d4d5544833ff6da4d79bc63587fd5ca
Current commit (broken): https://cgit.freedesktop.org/xorg/driver/xf86-video-intel/commit/?id=714052fb32a3c521efb295f2c400a673d515fb77

On the bright side, there are only 7 commits between those two commits, so it shouldn't be too hard to track down the cause.

aminb, 2016-11-15:
@patjak Any thoughts?

P.S. it's still broken as of this commit: https://cgit.freedesktop.org/xorg/driver/xf86-video-intel/commit/?id=dad64e9f76c718033402be7bfd2129866d492304

taylorthurlow, 2016-12-18:
Still broken it looks like.

taylorthurlow, 2016-12-18:
I found a workaround. Because the bug is an issue with `xf86-video-intel`, I did some digging and found [this discussion on Reddit](https://www.reddit.com/r/archlinux/comments/4cojj9/it_is_probably_time_to_ditch_xf86videointel/), where people suggested removing `xf86-video-intel` entirely. I had no issues doing so, with the exception of the `xbacklight` function no longer working.

Here is what I did as a workaround: I installed `mba6x_bl` as normal, then installed `light-git` from AUR. I can now set backlight as normal using `light -s mba6x_backlight -S 20` where `20` is the backlight percentage.