patjak/mba6x_bl #59: Won't build from Deb on MBA 7.1

simon-brooke, 2016-10-31:
Issues #45 and #55 relate, but I think this is not identical.

I'm trying to build on a MacBookAir7,1, which is of course not identical hardware, but which suffers from the same screen brightness problem. I don't believe this build fail is hardware related.

Installing gives the following:

    simon@jeweller:~/Downloads$ sudo dpkg -i mba6xbl-dkms_0.0.3_all.deb 
    (Reading database ... 333146 files and directories currently installed.)
    Preparing to unpack mba6xbl-dkms_0.0.3_all.deb ...

    ------------------------------
    Deleting module version: 0.0.3
    completely from the DKMS tree.
    ------------------------------
    Done.
    Unpacking mba6xbl-dkms (0.0.3) over (0.0.3) ...
    Setting up mba6xbl-dkms (0.0.3) ...
    Loading new mba6xbl-0.0.3 DKMS files...
    First Installation: checking all kernels...
    Building only for 4.4.0-45-generic
    Building for architecture x86_64
    Building initial module for 4.4.0-45-generic
    Error! Bad return status for module build on kernel: 4.4.0-45-generic (x86_64)
    Consult /var/lib/dkms/mba6xbl/0.0.3/build/make.log for more information.

Make log is attached.
[make.log.zip](https://github.com/patjak/mba6x_bl/files/562648/make.log.zip)

simon-brooke, 2016-10-31:
Please advise if there's anything I can do to help diagnose this problem.

patjak, 2016-10-31:
Looks like the deb package is outdated. Try building from source instead.

simon-brooke, 2016-11-01:
Thanks, will do.