patjak/mba6x_bl #63: Works on macbook pro 8,3

Daniel-Tyler-Farmer, 2017-02-09:
Just wanted to let you know that also works on the 2011 MacBook pro 8,3. Please update in supported hardware.

patjak, 2017-02-09:
Thanks. Do I understand it correctly that the 8,3 _requires_ mba6x_bl for suspend/resume and doesn't work properly without it?

Daniel-Tyler-Farmer, 2017-02-10:
On some distributions, they have either taken the time to get it working out of the box or are only using the intel graphics, but on others like Arch or Solus where the screen brightness doesn't work and you want to use the ati graphics, this will enable screen brightness. BTW thank you for making this, I have been trying to get screen brightness to work on here for the longest time.