patjak/mba6x_bl #66: mba6x_bl not working on kernel 4.10

sbng, 2017-07-07:
I just upgraded to kernel  version 4.10 and realize mba6x_bl module is causing problem to the suspend and resume operation. I removed the module to resolve the suspend/resume issue. Strangely my system now works on kernel 4.10 without mba6x_bl. Is this the expected behavior? Feeling pretty strange living without mba6x_bl after all these years. Am I missing something obvious?

patjak, 2017-07-08:
The bug was fixed in 4.8.5 (iirc) so mba6x_bl is actually no longer needed. mba6x_bl still does a better job at giving a linear brightness curve but not sure it's worth the hassle to keep the driver installed only for that.

sbng, 2017-07-09:
Thanks for the information. I really appreciate your work on mba6x_bl. Without your effort, my macbook air would has been quite un-usable in the past two years. Thanks for the support!

patjak, 2017-07-09:
you're welcome :)

closed 2017-07-09 by patjak

sbng, 2017-07-12:
Sorry for raising this issue again. Any chances u can port mba6x_bl to the newer kernel? been using the module from kernel 4.1.x and it hasn't been too reliable. I get close to 50% chance of unable to come back from suspend. Unlike the mba6x_bl module, I never fail to come back from suspend. Possible?