patjak/mba6x_bl #69: Build errors on kernel 6.5.0-10-generic

tr33, 2023-11-21:
I updated to ubuntu 23.10 which has kernel 6.5.0.

Building the module from git fails with

`# make
make -C /lib/modules/6.5.0-10-generic/build M=/usr/src/mba6x_bl modules
make[1]: Verzeichnis „/usr/src/linux-headers-6.5.0-10-generic“ wird betreten
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-4ubuntu3) 13.2.0
  CC [M]  /usr/src/mba6x_bl/mba6x_bl.o
/usr/src/mba6x_bl/mba6x_bl.c: In function ‘platform_probe’:
/usr/src/mba6x_bl/mba6x_bl.c:325:9: error: implicit declaration of function ‘acpi_video_set_dmi_backlight_type’; did you mean ‘acpi_video_get_backlight_type’? [-Werror=implicit-function-declaration]
  325 |         acpi_video_set_dmi_backlight_type(acpi_backlight_vendor);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         acpi_video_get_backlight_type
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:251: /usr/src/mba6x_bl/mba6x_bl.o] Fehler 1
make[2]: *** [/usr/src/linux-headers-6.5.0-10-generic/Makefile:2037: /usr/src/mba6x_bl] Fehler 2
make[1]: *** [Makefile:234: __sub-make] Fehler 2
make[1]: Verzeichnis „/usr/src/linux-headers-6.5.0-10-generic“ wird verlassen
make: *** [Makefile:8: all] Fehler 2
`

is the driver still required, or did it go upstream?

patjak, 2023-11-21:
The driver was intended to work around a bug in the i915 driver. The bug was eventually fixed so this driver is no longer needed.

It does, however give you a much more linear brightness curve so I kept using it for a while. But I haven't touch the machine for quite a few years so I have no idea if it still works. I know nothing about the Ubuntu package but it looks like the kernel API has changed.

patjak, 2023-11-21:
Then that's a different issue all together. Is i915 loading properly?

tr33, 2023-11-24:
It look like its being loaded:

`# lsmod|grep i915
i915                 4145152  41
drm_buddy              20480  1 i915
i2c_algo_bit           16384  1 i915
ttm                   110592  1 i915
drm_display_helper    241664  1 i915
cec                    94208  2 drm_display_helper,i915
drm_kms_helper        270336  2 drm_display_helper,i915
drm                   761856  44 drm_kms_helper,drm_display_helper,drm_buddy,i915,ttm
video                  73728  2 apple_bl,i915
`
dmesg output:
`
# dmesg|grep i915
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-6.5.0-10-generic root=/dev/mapper/vgubuntu-root ro crashkernel=384M-:128M i915.enable_dpcd_backlight=1 acpi_backlight=vendor libata.force=1:noncq intel_idle.max_cstate=1 pcie_aspm=force i915.enable_psr=0 i915.i915_enable_rc6=1 i915.i915_enable_fbc=1 i915.lvds_downclock=1 kopt=root=/dev/mapper/vgubuntu-root resume=/dev/mapper/vgubuntu-swap quiet splash vt.handoff=7
[    0.041715] Kernel command line: BOOT_IMAGE=/vmlinuz-6.5.0-10-generic root=/dev/mapper/vgubuntu-root ro crashkernel=384M-:128M i915.enable_dpcd_backlight=1 acpi_backlight=vendor libata.force=1:noncq intel_idle.max_cstate=1 pcie_aspm=force i915.enable_psr=0 i915.i915_enable_rc6=1 i915.i915_enable_fbc=1 i915.lvds_downclock=1 kopt=root=/dev/mapper/vgubuntu-root resume=/dev/mapper/vgubuntu-swap quiet splash vt.handoff=7
[    3.176906] i915: unknown parameter 'i915_enable_rc6' ignored
[    3.176909] i915: unknown parameter 'i915_enable_fbc' ignored
[    3.176911] i915: unknown parameter 'lvds_downclock' ignored
[    3.178146] i915 0000:00:02.0: vgaarb: deactivate vga console
[    3.179177] i915 0000:00:02.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=io+mem:owns=io+mem
[    3.241904] i915 0000:00:02.0: [drm] Skipping intel_backlight registration
[    3.242479] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
[    3.513817] fbcon: i915drmfb (fb0) is primary device
[    4.945114] i915 0000:00:02.0: [drm] fb0: i915drmfb frame buffer device
[   17.150192] snd_hda_intel 0000:00:03.0: bound 0000:00:02.0 (ops i915_audio_component_bind_ops [i915])
`

patjak, 2023-11-24:
Why are you passing all those kernel parameters to the i915 driver?

[ 3.241904] i915 0000:00:02.0: [drm] Skipping intel_backlight registration
The above line tells you that the i915 driver is skipping it's backlight registration.

Try removing:
i915.enable_dpcd_backlight=1
acpi_backlight=vendor

And you can also remove the following since they are not being used: 
[ 3.176906] i915: unknown parameter 'i915_enable_rc6' ignored
[ 3.176909] i915: unknown parameter 'i915_enable_fbc' ignored
[ 3.176911] i915: unknown parameter 'lvds_downclock' ignored

tr33, 2023-11-24:
The parameters are there for historical reasons: someone previously mentioned that these parameters fixed some problems - since then I have never read again whether they are still necessary. Because the machine was running pretty fine 🤷🏻‍♂️

removed both parameters -> works! AWESOME 💋😘

how about the other parameters - are they still necessary?

patjak, 2023-11-27:
If the machine is running fine I wouldn't bother. It's been to long since I used my MBA 6.2 to remember which quirks was actually useful. I'll close this issue now. Thanks.

closed 2023-11-27 by patjak